SilverFire/dessmonitor-homeassistant #1: PowMr POW-HVM2H-12V-N

lawyerhome, 2024-12-19:
I installed the integration in HAOS, according to the instructions (I repeated all the points) and tried to get telemetry of my inverter PowMr (POW-HVM2H-12V-N) in the card. Everything worked, and the card appeared, but it only displays all the data on the way from the inverter to the battery. Data (voltage, load) on the way from the Greed or to Load are displayed as zeros.
[image]
[image]

My Attributes of sensor.inverter_data are:

gd_: 
- id: gd_ac_input_frequency
  par: AC Input Frequency
  val: '49.9'
  unit: Hz
- id: gd_mains_status
  par: Mains Status
  val: Mains Discharge
- id: gd_ac_input_voltage
  par: AC Input Voltage
  val: '231.2'
  unit: V
- id: gd_ac_input_range
  par: AC Input Range
  val: Appliance

sy_: 
- id: sy_status
  par: Working State
  val: Line Mode
- id: sy_nonimal_output_apparent_power
  par: Nonimal Output Apparent Power
  val: '1600'
  unit: VA
- id: sy_nominal_output_voltage
  par: Nominal Output Voltage
  val: '230'
  unit: V
- id: sy_nominal_output_frequency
  par: Nominal Output Frequency
  val: '50.0'
  unit: Hz
- id: sy_main_cpu_version
  par: Main CPU Version
  val: '43969'
- id: sy_nonimal_output_active_power
  par: Nonimal Output Active Power
  val: '1600'
  unit: W
- id: sy_nominal_output_current
  par: Nominal Output Current
  val: '6'
  unit: A
- id: sy_rated_battery_voltage
  par: Rated Battery Voltage
  val: '12.0'
  unit: V
- id: sy_nominal_ac_current
  par: Nominal AC Current
  val: '6'
  unit: A
- id: sy_secondary_cpu_version
  par: Secondary CPU Version
  val: '0'
- id: sy_nominal_ac_voltage
  par: Nominal AC Voltage
  val: '230'
  unit: V
- id: sy_battery_piece
  par: Battery Piece
  val: 24V(3KW)
- id: sy_machine_type
  par: Machine Type
  val: Energy Storage Inverter

pv_: 
- id: pv_status
  par: PV Status
  val: PV Undervoltage
- id: pv_output_power
  par: PV Input Power
  val: '0'
  unit: W
- id: pv_input_voltage
  par: PV Input Voltage
  val: '8.0'
  unit: V

bt_: 
- id: bt_utility_charge
  par: Max Utility Charge Current
  val: '20'
  unit: A
- id: bt_battery_equalization_voltage
  par: Battery Equalization Voltage
  val: '14.4'
  unit: V
- id: bt_battery_charging_current
  par: Battery Charging Current
  val: '0'
  unit: A
- id: bt_vulk_charging_voltage
  par: Bulk Charging Voltage
  val: '14.1'
  unit: V
- id: bt_battery_equalization_interval
  par: Battery Equalization Interval
  val: '30'
  unit: day
- id: bt_battery_type
  par: Battery Type
  val: User
- id: bt_battery_status
  par: Battery Status
  val: Charging Batteries
- id: bt_battery_equalized_time
  par: Battery Equalized Time
  val: '60'
  unit: min
- id: bt_comeback_utility_iode
  par: Comeback Utility Mode Voltage Point Under (SBU Priority)
  val: '11.6'
  unit: V
- id: bt_battery_capacity
  par: Battery Capacity
  val: '100'
  unit: '%'
- id: bt_battery_discharge_current
  par: Battery Discharge Current
  val: '0'
  unit: A
- id: bt_floating_charging_voltage
  par: Floating Charging Voltage
  val: '13.5'
  unit: V
- id: bt_battery_voltage
  par: Battery Voltage
  val: '13.5'
  unit: V
- id: bt_battery_mode_voltage
  par: Comeback Battery Mode Voltage Point Under (SBU Priority)
  val: '13.5'
  unit: V
- id: bt_battery_equalized_timeout
  par: Battery Equalized Timeout
  val: '120'
  unit: min
- id: bt_battery_cut_off_voltage
  par: Low Battery Cut-off Voltage
  val: '12.0'
  unit: V
- id: bt_total_charge_current
  par: Max Total Charge Current
  val: '20'
  unit: A
- id: bt_charger_source_priority
  par: Charger Source Priority
  val: Solar only

bc_: 
- id: bc_output_apparent_power
  par: Output Apparent Power
  val: '94'
  unit: VA
- id: bc_output_voltage
  par: Output Voltage
  val: '231.2'
  unit: V
- id: bc_load_status
  par: Load Status
  val: Load On (Normal)
- id: bc_output_frequency
  par: Output Frequency
  val: '49.9'
  unit: Hz
- id: bc_output_source_priority
  par: Output Source Priority
  val: Utility
- id: bc_battery_capacity
  par: AC Output Load
  val: '4'
  unit: '%'

friendly_name: Inverter Data

What did I do wrong? Does anyone have any ideas? I also noticed that the current values ​​that the inverter PowMr transmits to the cloud (server) of dessmonitor (smartESS) are overstated by 5 Amperes (about), and this is not only for me, but also for other inverter PowMr users. The inverter PowMr display shows a correct charging current of 20A, but in the dessmonitor as 25A. That is bag.

SilverFire, 2024-12-19:
I've tested with Sorotec inverters, but PowMR sends a bit different attributes:

For example, Sorotec sends **Load Power** as `bc_load_active_power` attribute, measured in watts, so I can use it directly:

https://github.com/SilverFire/dessmonitor-homeassistant/blob/27bad9fe4480656df96159179e83209b1c633a57/src/template.yaml#L129-L135

PowMR sends the reactive power measured in Volt*Amperes in `bc_output_apparent_power` property, so you should convert in to active power and consider the [Power Factor](https://en.wikipedia.org/wiki/Power_factor), which is typically around 0.95 for the appartments.

```
- id: bc_output_apparent_power
  par: Output Apparent Power
  val: '94'
  unit: VA
```

So you should modify the load power state as follows:

```
    - name: "Load Power"
      state: >
        {% set bc_items = state_attr('sensor.inverter_data', 'bc_') %}
        {% set load_va = (bc_items | selectattr('id', 'equalto', 'bc_load_active_power') | map(attribute='val') | first) | float %}
        {{ load_va * 0.95}}
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"
```

Check other properties from the [template.yaml](https://github.com/SilverFire/dessmonitor-homeassistant/blob/main/src/template.yaml#L129-L135) to alight them with your inverter data.

lawyerhome, 2024-12-19:
I made these changes to the /homeassistant/template.yaml file (replacing lines 129-136), but it didn't work. 
The problem is not in the incorrect data that is displayed in VA instead of W, but in the fact that some entities are not seen by HA (their status is "unavailable") at all, for example:
[image]
[image]
All data (voltage, current, load) from Greed, Solar Panels and Load data is not available. 
Also I can't add sensor.inverter_data to the Energy panel

lawyerhome, 2024-12-20:
Thank you for your help. 
Indeed, the thing is that the entity names (ID) that you received from API and write in tamplate.yaml file are slightly different from those that I receive from the API for my inverter. For example: 
1.) your variant **bc_load_active_power** but my variant **bc_output_apparent_power**.  
2.) your variant **gd_grid_voltage** but my variant **gb_ac_input_voltage**
3.) youre variant **gd_rid_frequency** but my variant **gb_ac_input_frequency**
As a result of the replacement, I was able to see the voltage (231,1V) and energy consumption (89W)! 
[image]

lawyerhome, 2024-12-20:
I got this result. 
[image]
The energy consumption data appeared, but if the consumption data from the Greed corresponds to reality, then the data on consumption from the inverter does not correspond to reality and is incorrect.
My inverter consumes about 31 W at idle, so, taking 93 W from the Greed, it takes 31 W, and transmits only 62 W further to the house.
This is also indicated by the load on the inverter 4% (bc_battery_capacity)
[image]
of its nominal (1600 W) - 64 W (sy_nominal_output_active_power). 
[image]
How can I write a formula in the template file so that the data is displayed correctly? From Greed to Inverter - 93W, and from Inverter to House - 64W?

lawyerhome, 2024-12-20:
Thats it!
[image]
And my result
[image]
`    - name: "Load Power"
      state: >
        {% set bc_items = state_attr('sensor.inverter_data', 'bc_') %}
        {% set load_percent = (bc_items | selectattr('id', 'equalto', 'bc_battery_capacity') | map(attribute='val') | first) | float %}
        {{ 1600 * load_percent / 100}}
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"`

SilverFire, 2024-12-20:
I'm glad to hear about your success!

Could you create a PR to add a `src/powmr.yaml` file with the correct configuration for others to reuse?

> {{ ((charging_current * battery_voltage) + load_power) * 0.95 }}

I recommend making the following change:

```
{{ ((charging_current * battery_voltage) + (load_power * 0.95)) }}
```

This adjustment will enhance precision, as the battery charging current is not a reactive power and therefore does not require conversion.

lawyerhome, 2024-12-20:
[image]
I am so sorry, I do not knaw how to create PR there. But I can attach file there: /homeassistant/template.yaml and code of card.
[powmr_yaml.txt](https://github.com/user-attachments/files/18211716/powmr_yaml.txt)
[type customsunsynk-power-flow-card.txt](https://github.com/user-attachments/files/18211718/type.customsunsynk-power-flow-card.txt)

SilverFire, 2024-12-20:
Added and updated manuals, thank you.

andrzejano, 2025-03-01:
please for easun inverter

SilverFire, 2025-03-02:
> please for easun inverter

Hi. Please, create a separate issue for that.

andrzejano, 2025-03-02:
I have no experience and I am asking for help

carlesdorado, 2025-03-17:
Would it also be possible to integrate the Voltronic Axpert Max II 8Kw inverter?
Some parameters work (battery data). Others are missing...
I'd appreciate your help. I haven't found anything reliable online except your work. Thanks!

commit dab34c9d (2024-12-20): Added PowMr template

Closes #1