SilverFire/dessmonitor-homeassistant #4: Automatically weekly update dessmonitor_api_uri

lawyerhome, 2025-01-09:
Every week the sensors of my PowMr invertor stop receiving data from the API URL (they get an error), and zeros appear on the map.
[image]
If manual put fresh API URL on secrets.yaml (dessmonitor_api_uri) - it brings results, data appears, but only on one week, after which you need to update the API URL again. 
Is it possible to automatically receive a fresh URL (with a new token) and replace it in secrets.yaml?

SilverFire, 2025-01-09:
Hi!
There is a neighbor issue #3 with exampe, that calculates `sign` the same way the application does it.
I did not test it though. If you can try it and confirm it works – that would be great!

SilverFire, 2025-01-14:
We discuss this idea in #3, feel free to join a discussion there