SilverFire/dessmonitor-homeassistant #6: Dessmonitor - JSDSOLAR

picasso-12345, 2025-01-27:
Hi, 
I have a JSDSOLAR 5.5kW 48V solar inverter. It is registered in Dessmonitor, I get regular readings. 
When I enter:
dessmonitor_api_uri: https://web.dessmonitor.com/public/?sign=08acf0c4247bb5e6       .........    &i18n=en_US

I will get an answer:
{"err":0,"desc":"ERR_NONE","dat":{"gts":"2025-01-27 18:02:43","pars":{"gd_":[{"id":"gd_grid_voltage","par":"Mains input voltage","val":"235.1","unit":"V"},{"id":"gd_bse_grid_rating_frequency_read","par":"Mains frequency","val":"50.03","unit":"Hz"}],"pv_":[{"id":"pv_output_power","par":"PV Power","val":"0","unit":"W"},{"id":"pv_input_current","par":"PV current","val":"0.00","unit":"A"},{"id":"pv_voltage","par":"PV voltage","val":"0.0","unit":"V"}],"sy_":[{"id":"sy_status","par":"Operating mode","val":"Mains mode"}],"bt_":[{"id":"bt_model","par":"DCDC control status","val":"Charge"},{"id":"bt_battery_capacity","par":"Battery capacity","val":"100","unit":"%"}],"bc_":[{"id":"bc_load_percent","par":"Load percentage","val":"10","unit":"%"}]}}}

1) What answer do you get from the Dessmonitor app, please?
2) How to find out, e.g. battery voltage when i see this value in dessmonitor app?

Thank you

SilverFire, 2025-01-29:
Hello.

> I will get an answer:

This is a correct API response.

> What answer do you get from the Dessmonitor app, please?

The amount of data depends on the inverter model. Some models provide [more detailed data](https://github.com/SilverFire/dessmonitor-homeassistant/issues/1#issue-2750701639), while your invertor provide only the basics.

> How to find out, e.g. battery voltage when i see this value in dessmonitor app?

If you see the voltage in the UI, it should be in on of the API responses. When you grab the credentials (item 4 in the [Preparation](https://github.com/SilverFire/dessmonitor-homeassistant?tab=readme-ov-file#preparation) article), try to check other responses and look for the number you see in UI.

lontk, 2025-01-29:
> 2\. How to find out, e.g. battery voltage when i see this value in dessmonitor app?

I have a Must inverter and there is also no battery voltage. It depends on the inverter. Also, I'd like to point out that the dessmonitor_api_uri link changes weekly and you need to update it. Hopefully, the author will figure out how to fix it.

picasso-12345, 2025-01-30:
This data can be read in Dessmonitor:

{"err":0,"desc":"ERR_NONE","dat":{"parameter":[{"par":"battery_voltage","name":"Battery voltage","val":"57.2000","unit":"V"},{"par":"bt_battery_capacity","name":"Battery capacity","val":"100.0000","unit":"%"},{"par":"gd_grid_voltage","name":"Mains input voltage","val":"234.4000","unit":"V"},{"par":"output_current_percentage","name":"Output current percentage","val":"11.0000","unit":"%"}]}}

The battery voltage value is indicated there. How can I get this value, please? This write up doesn't work for me.
configuration.yaml   is:
sensor:
  - platform: rest
    name: Inverter Data1
    resource_template: !secret dessmonitor_api_ura
    method: GET
    json_attributes_path: "$.dat.parameter"
    json_attributes:
      - battery_    
    scan_interval: 60 # Update every 1 minutes
    value_template: "OK"
template: !include template1.yaml

template1.yaml   is:
sensor:
   - name: "Battery Voltage"
      state: >
        {% set battery_items = state_attr('sensor.inverter_data1', 'battery_') %}
        {{ (battery_items | selectattr('par', 'equalto', 'battery_voltage') | map(attribute='val') | first) | float }}
      unit_of_measurement: "V"
      device_class: "voltage"
      state_class: "measurement"

Where is the error? Please.

picasso-12345, 2025-02-25:
> Komu: SilverFire/dessmonitor-homeassistant <dessmonitor-homeassistant@
> noreply.github.com>
> com>
> (Issue #6)
> "
> Hi.
> I'm trying to integrate my inverter Must into HA. I'm new with HA. I've
> already registered in dessmonitor website. I don't understand where I need
> to enter url from that site. Would you explain details what integration I
> need to use and how to add my inverter to my HA? Please.
>
> (https://github.com/SilverFire/dessmonitor-homeassistant/issues/6#issuecomment-2681268572)
>
> (https://github.com/SilverFire/dessmonitor-homeassistant/issues/6#issuecomment-2681268572)
>
> Hi.
> I'm trying to integrate my inverter Must into HA. I'm new with HA. I've
> already registered in dessmonitor website. I don't understand where I need
> to enter url from that site. Would you explain details what integration I
> need to use and how to add my inverter to my HA? Please.
>
> (https://github.com/SilverFire/dessmonitor-homeassistant/issues/6#issuecomment-2681268572)
> "

Hi, I'm happy to help. Let's do it

1. Install RESTfull integration in Home Assistant (HA). Settings - Devices 
and Services - +Add Integration - RESTful

2. File Editor - file secrets.yaml, here on the first line insert 

dessmonitor_api_uri: https://web.dessmonitor.com/public/..... + more from 
your Dessmonitor. 

Don't forget to save the file!

3. File Editor - file configuration.yaml, here insert:

# Dessmonitor

sensor:

  - platform: rest

    name: Inverter Data

    scan_interval: 120 # Update every 2 minutes

    value_template: "OK"

    resource_template: !secret dessmonitor_api_uri   # viz file secrets.yaml

    method: GET

    json_attributes_path: "$.dat.pars"

    json_attributes:

      - gd_

      - sy_

      - pv_

      - bt_

      - bc_

Don't forget to save the file!

4. File Editor - sensor.yaml file, here insert:

# Střídač(měnič) ***********************************************************
********

    - name: "Grid Voltage"

      state: >

        {% set gd_items = state_attr('sensor.inverter_data', 'gd_') %}

        {{ (gd_items | selectattr('id', 'equalto', 'gd_grid_voltage') | map
(attribute='val') | first) | float }}

      unit_of_measurement: "V"

      device_class: "voltage"

      state_class: "measurement"

    - name: "Frequency grid"

      state: >

        {% set gd_items = state_attr('sensor.inverter_data', 'gd_') %}

        {{ (gd_items | selectattr('id', 'equalto', 'gd_bse_grid_rating_
frequency_read') | map(attribute='val') | first) | float }}

      unit_of_measurement: "Hz"

      device_class: "frequency"

      state_class: "measurement"

    - name: "Grid Power"

      state: >

        {% set gd_items = state_attr('sensor.inverter_data', 'gd_') %}

        {% set bt_items = state_attr('sensor.inverter_data', 'bt_') %}

        {% set bc_items = state_attr('sensor.inverter_data', 'bc_') %}

        {% set grid_voltage = (gd_items | selectattr('id', 'equalto', 'gd_
grid_voltage') | map(attribute='val') | first) | float %}

        {% set charging_current = (bt_items | selectattr('id', 'equalto', 
'bt_battery_charging_current') | map(attribute='val') | first) | float %}

        {% set battery_voltage = (bt_items | selectattr('id', 'equalto', 'bt
_battery_voltage') | map(attribute='val') | first) | float %}

        {% set load_power = (bc_items | selectattr('id', 'equalto', 'bc_load
_active_power') | map(attribute='val') | first) | float %}

        

        {% if grid_voltage > 0 %}

          {{ (charging_current * battery_voltage) + load_power }}

        {% else %}

          0

        {% endif %}

      unit_of_measurement: "W"

      device_class: "power"

      state_class: "measurement"

    - name: "PV1 Voltage"

      state: >

        {% set pv_items = state_attr('sensor.inverter_data', 'pv_') %}

        {{ (pv_items | selectattr('id', 'equalto', 'pv_voltage') | map
(attribute='val') | first) | float }}

      unit_of_measurement: "V"

      device_class: "voltage"

      state_class: "measurement"

    - name: "PV1 Current"

      state: >

        {% set pv_items = state_attr('sensor.inverter_data', 'pv_') %}

        {{ (pv_items | selectattr('id', 'equalto', 'pv_input_current') | map
(attribute='val') | first) | float }}

      unit_of_measurement: "A"

      device_class: "current"

      state_class: "measurement"

    - name: "PV1 Power"

      state: >

        {% set pv_items = state_attr('sensor.inverter_data', 'pv_') %}

        {{ (pv_items | selectattr('id', 'equalto', 'pv_output_power') | map
(attribute='val') | first) | float }}

      unit_of_measurement: "W"

      device_class: "power"

      state_class: "measurement"

   

.

.

.

Don't forget to save the file!

5. Restart HA. And that's it.

Zbynek

---------- Původní e-mail ----------

closed 2025-03-02 by SilverFire