SilverFire/dessmonitor-homeassistant #7: Daxtrom inverters

Xcpro1, 2025-03-16:
Hi, I tried with my Daxtrom 6,2kw (NM-IV-6,2-PLUS) and no data is available. How and where can I find the correct API? Thank you in advance.

SilverFire, 2025-03-24:
Hi!

First, confirm if your inverter works with the SmartESS app:
1. Install the SmartESS app on your smartphone
2. Try connecting your inverter to the app
3. If successful, register and check if you can view your data at dessmonitor.com

If you can see your inverter data on dessmonitor.com, you can follow the integration steps from the [repository documentation](https://github.com/SilverFire/dessmonitor-homeassistant?tab=readme-ov-file#preparation).

If not – then this ivertor is not supported by this API integration.

Xcpro1, 2025-03-25:
Update, not all sensors are unavailable. Load Power, frequency, voltage and all sensors of PV side is missing. Battery sensors, temp. and status are monitoring for example...

Xcpro1, 2025-03-26:
Another update, trying different templates I found that the PowMr one show all important info, only PV current Is missing, but volt and power not...  🙂

SilverFire, 2025-04-02:
Could you show the JSON, you get from the `dessmonitor_api_uri`?