SilverFire/dessmonitor-homeassistant #8: Easun smgIII 4.2kW

andrzejano, 2025-03-17:
I am asking for help in integrating the easun smgIII 4.2kW ​​inverter with home assistant either via data from dessmonitor or using esphome esp32 and rs232 ttl converter. So far I have not been able to get it to work.

SilverFire, 2025-03-24:
Hi!

First, confirm if your inverter works with the SmartESS app:
1. Install the SmartESS app on your smartphone
2. Try connecting your inverter to the app
3. If successful, register and check if you can view your data at dessmonitor.com

If you can see your inverter data on dessmonitor.com, you can follow the integration steps from the [repository documentation](https://github.com/SilverFire/dessmonitor-homeassistant?tab=readme-ov-file#preparation).

andrzejano, 2025-03-25:
what sensor is responsible for battery animation in sunsync power card

SilverFire, 2025-04-02:
I believe the `Battery Current` or `Battery Power`: positive for charging and negative for discharging. Or wise-versa.