SilverFire/dessmonitor-homeassistant #12: POW-HVM10.2M Incorrect Values

SpankyDevil, 2025-03-27:
Hi, Thank you for your effort. This is great work.

Can I please get some assistance correcting some values.

My inverter is a POW-HVM10.2M

[image]

[image]

Is it also possible to add the values that are not already included?

Is there a way to configure the Inverter through Home Assistant, the same as you can through Dessmonitor (but better lol)

SilverFire, 2025-04-01:
Hi there.

> Can I please get some assistance correcting some values.

The "PV1 current" state shows as "Unavailable" because this property isn't included in the Dessmonitor API response data you provided. However, your inverter does supply `pv_power` and `pv_input_voltage`.

Since `Current = Power / Voltage`, you can calculate this value yourself. You can find an example of how to implement this in the [template configuration](https://github.com/SilverFire/dessmonitor-homeassistant/blob/main/src/template.yaml#L18-L36).

> Is it also possible to add the values that are not already included?

Yes, you can add the missing values. Please follow the structure outlined in the [template.yaml](https://github.com/SilverFire/dessmonitor-homeassistant/blob/main/src/template.yaml) file and include the additional properties you receive from the API.

> Is there a way to configure the Inverter through Home Assistant, the same as you can through Dessmonitor (but better lol)

Regarding configuring the inverter via Home Assistant, this is technically possible, but not implemented in this integration. Please follow the discussion in issue #10 for more details and context.

closed 2025-04-01 by SilverFire

SpankyDevil, 2025-04-01:
Thanks for your reply.

Sorry, I should have been more precise regarding correcting some values.

Load Power is reading a little amount compared to the correct value

Is Battery Current & Battery Power supposed to read a negative value while it's being charged and not in use?

[image]

[image]

SilverFire, 2025-04-02:
Try changing `bc_load_active_power` to `bc_output_apparent_power` in the [Load Power](https://github.com/SilverFire/dessmonitor-homeassistant/blob/main/src/template.yaml#L129-L136) sensor configuration.

SpankyDevil, 2025-04-02:
> Try changing `bc_load_active_power` to `bc_output_apparent_power` in the [Load Power](https://github.com/SilverFire/dessmonitor-homeassistant/blob/main/src/template.yaml#L129-L136) sensor configuration.

I'm using the PowMR Template, which has this configured.

[image]

SpankyDevil, 2025-04-02:
Changed to your suggestion, it looks to be working now.

Thank you. 

[image]

SilverFire, 2025-04-02:
Great job!

SpankyDevil, 2025-04-02:
Has anyone managed to get the Daily amounts working?

SilverFire, 2025-04-02:
Check out https://www.home-assistant.io/integrations/utility_meter/