SilverFire/dessmonitor-homeassistant #13: Was working last week now I'm getting error {"err":10,"desc":"ERR_NO_AUTH"}

SpankyDevil, 2025-04-07:
Was working last week now I'm getting error {"err":10,"desc":"ERR_NO_AUTH"}

darth-romios, 2025-04-10:
+1
looks like they added some protection for this request :-(

SilverFire, 2025-04-14:
Hi.
I'm not able to test right away.
Does the Dessmonitor UI work for you?

hunterelectros, 2025-04-26:
Hello! Most likely the cause is that copied URL is not valid after 1 week. Every week need to reinit authorization creds or every week copy new, valid URL. For example, here https://github.com/theLarryds/solar , author did script that send correct request every time. But as for me, from all variables I cannot figure out with companyKey option. I canot find it anywhere and understand where I can get it.

****
//https://api.dessmonitor.com/chapter1/apiHelp.html
// Define your parameters here
// "usr": your DESS Monitor username - used to sign into the app,
// "pwd": password from the Dess site,
// "**companyKey**": This I derived from the network inspector tool in the browser to find the "companyKey" attribute,
// "source": "1" - App per the API Docs example,
// "appClient": "android" - App per the API Docs example,
// "appId": "com.demo.test" - App per the API Docs example,
// "appVersion": "3.6.2.1" - App per the API Docs example,
****

hunterelectros, 2025-04-26:
Huh. I tried some actions and found my company-key )). Do next:
Login with your creds in https://www.dessmonitor.com. At the same time open debug browser tool (Ctrl+Shift+I). Chose Network and type in search field "company-key". The founded request will have company-key value :)

[image]